ciaa/Hardware #57: CIAA-FSL and CIAA-NXP: Input PTC should be replaced.

nscotti, 2015-06-02:
MF-SM300 (TH1101 in CIAA-FSL, TH1 in CIAA-NXP) should be replaced by another PTC because of the VDC parameter. VDC should be greater than 24 V.
One option is use the Littelfuse part number **1812110/33** (http://www.littelfuse.com/~/media/electronics/datasheets/resettable_ptcs/littelfuse_ptc_1812l_datasheet.pdf.pdf). This part number is already used in CIAA-FSL board. **The footprint is smaller** but is the best option for not adding a new component.
Another option is use a Bourns part number with the same size of MF-SM300: **MF-SM100** (http://www.bourns.com/data/global/pdfs/mfsm.pdf).

pridolfi, 2015-06-02:
We need to check all PTCs, VDC parameter seems to be wrong. :-(

pridolfi, 2015-08-10:
TH1 recomendable: 2A 24V 2920
0ZCF0200FF2C
http://ar.mouser.com/ProductDetail/Bel-Fuse/0ZCF0200FF2C/?qs=sGAEpiMZZMsxR%252bBXi4wRUHZtKAyWcVPATxbIc5OCM0stlbWjfigW4g%3d%3d