ciaa/Hardware #59: Reset al tocar battery holder

mirkovich0, 2015-06-12:
Se resetea el microcontrolador (no necesariamente siempre) al tocar con la mano el zócalo de la batería (metálico, accesible desde afuera) si no se encuentra la batería presente (posiblemente poniendo la batería no suceda).

Se encuentra que el pin está conectado directo al micro, sin ningún filtrado.

cuencagonzalo, 2015-06-18:
Revisando los esquemáticos, veo que ambas NXP (CIAA y EDU-CIAA), están conectadas de la misma manera.  La CIAA-FSL tiene un capacitor de 100nF filtrando.

mabeett, 2016-11-08:
Confirmo problema con edu-ciaa-nxp A45570756.

mabeett, 2016-11-09:
Confirmo el problema en CIAA-NXP sin número de serie, de prelanzamiento, con soporte para batería y sin batería montada.
También en CIAA-NXP Número de código de barras CIAA0192, que tiene pila soldada. La pila tiene poca carga (0.9V en vez de 3) la inestabilidad con el ruido existe también.